vsergeev/luaradio #13: ffi path discovery OSX

akhenakh, 2016-07-03:
I'm using Macports in OSX so most of my libs are located in `/opt/local`
I have to load the shared liquid dsp lib by its full path for luaradio to detect it.
```
lib_available, lib = pcall(ffi.load, "/opt/local/lib/libliquid.dylib")
if lib_available then
  platform.libs["liquid"] = lib
  platform.features["liquid"] = true
end
```
I've tried LD_LIBRARY_PATH  to avoid patching with no luck, I'm not very familiar with luajit.
Thanks.

vsergeev, 2016-07-03:
I think in Mac OS X's case the equivalent environment variable is `DYLD_LIBRARY_PATH`. Another option might be to add `/opt/local/lib` to Mac OS X's equivalent of `/etc/ld.so.conf` and rerunning Mac OS X's equivalent of `sudo ldconfig`... Unfortunately, I don't have a Mac to test on.

akhenakh, 2016-07-03:
tried `DYLD_FALLBACK_LIBRARY_PATH` & `DYLD_LIBRARY_PATH` with no result.
Thanks for the tip anyway

rht, 2016-10-16:
You could manually set `LDFLAGS` to include `-L/opt/local/lib` and/or `CFLAGS`/`CPPFLAGS` to include `-I/opt/local/include`. Though this could be set in macports config as well.

jimla44, 2016-11-29:
You can make symbolic links: from /opt/local/lib to /usr/local/lib:
sudo ln -s /opt/local/lib/libvolk.dylib /usr/local/lib/libvolk.dylib
sudo ln -a /opt/local/lib/libliquid.dylib /usr/local/lib/libliquid.dylib

rfc6919, 2017-12-29:
This issue is the result of macOS stripping environment variables that control dyld from the environment whenever a System Integrity Protection protected binary is executed (see the dyld(1) manpage). In this case, the problem is LD_LIBRARY_PATH and DYLD_LIBRARY_PATH being stripped when /usr/bin/env is executed through the luaradio shebang.

I'm not sure of a good general solution here, but the way I'm working around it is changing the luaradio shebang from ``#!/usr/bin/env luajit`` to ``#!/opt/brew/bin/luajit`` so my LD_LIBRARY_PATH=/opt/brew/lib isn't stripped out by the extra exec via env(1).

A simple reduction of this problem is:

``XYZZY=foobar env | grep XYZZY`` => prints ``XYZZY=foobar``
``LD_LIBRARY_PATH=foobar env | grep LD_LIBRARY_PATH`` => prints nothing

or, assuming your libraries are in /opt/brew/lib, the actual ffi.load("liquid") failure:

``LD_LIBRARY_PATH=/opt/brew/lib luajit -e 'require("ffi").load("liquid")'`` => succeeds
``LD_LIBRARY_PATH=/opt/brew/lib env luajit -e 'require("ffi").load("liquid")'`` => error

vsergeev, 2020-03-23:
Package added to MacPorts with macports/macports-ports#6539 with a wrapper script to fix dynamic library loading in LuaRadio.

closed 2020-03-23 by vsergeev