vsergeev/luaradio #14: Connection refused on Mac OS X

mehdideveloper, 2016-07-03:
Hi
I just built the luaradio.
Here's its output when run with "--platform":
`luajit          LuaJIT 2.0.4`
`os              OSX`
`arch            x64`
`page size       4096`
`cpu count       4`
`cpu model       MacBookPro12,1`
`features`
`   fftw3f      true`
`  volk        true`
`    liquid      true`

Now when I want to start an example, it gives me "Connection refused" error and exits:
>Found Rafael Micro R820T tuner
>Exact sample rate is: 1102500.002464 Hz
>Using Volk machine: avx_64_mmx
>Using Volk machine: avx_64_mmx
>Using Volk machine: avx_64_mmx
>Using Volk machine: avx_64_mmx
>W: [] caps.c: Normally all extra capabilities would be dropped now, but that's impossible because PulseAudio was built without capabilities support.
>**[PulseAudioSink] Block runtime error: ./radio/blocks/sinks/pulseaudio.lua:95: pa_simple_new(): Connection refused**
>stack traceback:
>	[C]: in function 'error'
>	./radio/blocks/sinks/pulseaudio.lua:95: in function 'process'
>	./radio/core/block.lua:382: in function 'run'
>	./radio/core/composite.lua:554: in function <./radio/core/composite.lua:554>
>	[C]: in function 'xpcall'
>	./radio/core/composite.lua:554: in function 'start'
>	./radio/core/composite.lua:451: in function 'run'
>	examples/rtlsdr_wbfm_mono.lua:34: in main chunk
>	[C]: in function 'dofile'
>	./luaradio:105: in main chunk
>	[C]: at 0x0100002150
>Using Volk machine: avx_64_mmx
>Using Volk machine: avx_64_mmx`

Version of OS: 10.11.5

Any ideas?
Thanks

vsergeev, 2016-07-03:
Do you have the PulseAudio server running?

The PulseAudio audio sink was intended more for Linux systems, but you're welcome to try to run it on OS X.

mehdideveloper, 2016-07-03:
OK thanks. I installed pulseaudio (latest version: 8.0) and now it works and I have sound.
Another question: shouldn't I see a graph/diagram/spectrum as well?
(At first run, it failed because I didn't have gnuplot. I installed gnuplot, but it does not show anything. I'm running FM examples)
In the code, there are some GnuPlotSpectrumSink objects.

vsergeev, 2016-07-03:
Yup, the plotting should work too. Does Mac OS X use the `DISPLAY` variable? That's what the script checks is set to enable plotting. If not, you can try setting it with a dummy value when running an example: `DISPLAY=foo ./luaradio examples/rtlsdr_...`.

mehdideveloper, 2016-07-03:
I set "DISPLAY" environment variable, but it didn't make any difference (I can't still see a spectrum)
I uninstalled gnuplot and installed it again with "--x11" argument using brew.
This time it was just showing a static spectrum and freezed!, so I uninstalled it again and installed gnuplot with "--with-qt" argument; this time it showed multiple spectrum windows, but they were unusable (audio had glitches; starts and stops; I can't select any of those spectrum windows. It's like, the focus is changing rapidly from a window to another, so I had to remove dongle from my laptop to stop luaradio)

Why it ignores the "DISPLAY" in the first case? How should I troubleshoot the problem?

Thanks for your helps.

vsergeev, 2016-07-03:
The `DISPLAY` variable is just what the scripts checks to decide whether or not to connect the plotting sinks, so you can run the examples on a headless environment (e.g. over SSH) without plotting. The value isn't used in any way.

At the moment I'm not sure what the next best thing to try is (I don't have a Mac to test on). I've seen one comment of someone getting the plotting to work on OS X, but I'm not sure how their setup differs from yours.

mehdideveloper, 2016-07-04:
OK, I somehow fixed it.
I removed gnuplot and installed it again with brew, using the "--with-x11" switch. 
(You should use "--with-x11" if you wish to use XQuartz (X11) system for plotting)
I also set the DISPLAY to ":0.0"
Now it runs and shows the three spectrum windows, but they overlap on each other very fast (because of update time of 0.05 seconds)
If I comment out the code for displaying plot2 and plot3, and just show the plot1, everything's perfect.
So, I think you should draw the windows in different sections of display (so they won't overlap) or fix the refresh problem. It seems they're struggling to be shown on top of each other.

vsergeev, 2016-07-22:
I'm going to close this issue for now to consolidate the gnuplot issue discussion to #26.

closed 2016-07-22 by vsergeev