vsergeev/luaradio #16: RTLSDR default gain settings don't work with E4000 tuner

kevinmehall, 2016-07-03:
With my older RTLSDR, all the examples fail with:

```
Found Elonics E4000 tuner
[RtlSdrSource] Block runtime error: ./radio/blocks/sources/rtlsdr.lua:118: rtlsdr_set_tuner_gain(): -1
stack traceback:
	[C]: in function 'error'
	./radio/blocks/sources/rtlsdr.lua:118: in function 'initialize_rtlsdr'
	./radio/blocks/sources/rtlsdr.lua:171: in function 'run'
	./radio/core/composite.lua:554: in function <./radio/core/composite.lua:554>
	[C]: in function 'xpcall'
	./radio/core/composite.lua:554: in function 'start'
	./radio/core/composite.lua:451: in function 'run'
	examples/rtlsdr_wbfm_mono.lua:34: in main chunk
	[C]: in function 'dofile'
	./luaradio:105: in main chunk
	[C]: at 0x004044a0
```

I was able to work around this by setting the `autogain` option, or by setting `rf_gain` to a [supported value](http://cgit.osmocom.org/rtl-sdr/tree/include/rtl-sdr.h?id=master#n206):

```.diff
--- a/examples/rtlsdr_wbfm_mono.lua
+++ b/examples/rtlsdr_wbfm_mono.lua
@@ -9,7 +9,7 @@ local frequency = tonumber(arg[1])
 local tune_offset = -250e3
 
 -- Blocks
-local source = radio.RtlSdrSource(frequency + tune_offset, 1102500, {freq_correction = 1.0})
+local source = radio.RtlSdrSource(frequency + tune_offset, 1102500, {freq_correction = 1.0, autogain=true})
 local tuner = radio.TunerBlock(tune_offset, 200e3, 5)
 local fm_demod = radio.FrequencyDiscriminatorBlock(1.25)
 local af_filter = radio.LowpassFilterBlock(128, 15e3)
```

vsergeev, 2016-07-03:
I see, thanks. Looks like the source might have to check `rtlsdr_get_tuner_gains()` and pick a reasonable default.

vsergeev, 2016-07-03:
Interesting, my dongle (with R820T2 tuner) doesn't support the 10.0 dB default according to `rtlsdr_get_tuner_gains()`, but the gain set goes through without error.

commit 72d9896d (2016-07-03): blocks/sources/rtlsdr: choose default rf gain from dongle supported gains

closest to 10 dB, instead of assuming the dongle supports exactly 10 dB.
fixes #16.
commit d5ad831e (2016-07-04): blocks/sources/rtlsdr: choose default rf gain from dongle supported gains

closest to 10 dB, instead of assuming the dongle supports exactly 10 dB.
fixes #16.
commit 41cba45a (2016-07-04): blocks/sources/rtlsdr: choose default rf gain from dongle supported gains

closest to 10 dB, instead of assuming the dongle supports exactly 10 dB.
fixes #16.
commit 1673d374 (2016-07-04): blocks/sources/rtlsdr: choose default rf gain from dongle supported gains

closest to 10 dB, instead of assuming the dongle supports exactly 10 dB.
fixes #16.
closed 2016-07-04 by vsergeev