vsergeev/luaradio #18: Add HackRF source block

zeryl, 2016-07-04:
I know it's not the most popular, but would be awesome to have a HackRF block available.

vsergeev, 2016-07-04:
The more the merrier.

It's just a matter of wrapping `libhackrf`, once I get an asynchronous callback wrapper in place (which will come with implementing the Airspy source).

zeryl, 2016-07-04:
Really wish I knew enough to do so.  If there's anything I can do to help, as far as testing and such, let me know.  I am "comfortable" with LUA from WoW addon development, so think I can do something with this, or at least, I hope so!

vsergeev, 2016-07-04:
Testing would be super helpful, as I don't currently own a HackRF... I'll post here once I get something started.

vsergeev, 2016-07-09:
@zeryl, @dominicgs, or others, if you guys have a chance, the HackRF source block is ready for some testing.

It's available in the `devel` branch:  `git clone https://github.com/vsergeev/luaradio.git --branch devel`

One of the RTL-SDR examples, e.g. `rltsdr_wbfm_mono.lua`, can be modified to use it:

```
...
local source = radio.AirspySource(frequency + tune_offset, 10e6)
local tuner = radio.TunerBlock(tune_offset, 200e3, 40)
...
...
```

rxseger, 2016-07-09:
@vsergeev testing with devel as of https://github.com/vsergeev/luaradio/commit/7f226a0026823fe235ba8da9769f96b46ada2328 with a HackRF One and those changes to examples/rtlsdr_wbfm_mono.lua I get this error:

```
luaradio $ luaradio examples/rtlsdr_wbfm_mono.lua 89.9
[HackRFSource] Block runtime error: ./radio/blocks/sources/hackrf.lua:199: bad argument #1 to 'hackrf_error_name' (cannot convert 'number' to 'void')
stack traceback:
	[C]: in function 'hackrf_error_name'
	./radio/blocks/sources/hackrf.lua:199: in function 'initialize_hackrf'
	./radio/blocks/sources/hackrf.lua:242: in function 'run'
	./radio/core/composite.lua:554: in function <./radio/core/composite.lua:554>
	[C]: in function 'xpcall'
	./radio/core/composite.lua:554: in function 'start'
	./radio/core/composite.lua:451: in function 'run'
	examples/rtlsdr_wbfm_mono.lua:34: in main chunk
	[C]: in function 'dofile'
	/usr/local/bin/luaradio:105: in main chunk
	[C]: at 0x0100002150
```

vsergeev, 2016-07-09:
@rxseger, thanks. Try 272a84046ae32dac4765bd8a8a063a3a916643eb, it should render the error correctly. Also, remember to pass e6 for MHz in your argument (that's likely why it was complaining about setting the frequency).

zeryl, 2016-07-09:
@vsergeev I tried doing the rtlsdr_ax25.lua example, and get the following:

[HackRFSource] Block runtime error: ./radio/core/async.lua:106: lua_pcall(): ./radio/blocks/sources/hackrf.lua:205: module 'radio' not found:
        no field package.preload['radio']
        no file './radio.lua'
        no file '/usr/share/luajit-2.0.4/radio.lua'
        no file '/usr/local/share/lua/5.1/radio.lua'
        no file '/usr/local/share/lua/5.1/radio/init.lua'
        no file '/usr/share/lua/5.1/radio.lua'
        no file '/usr/share/lua/5.1/radio/init.lua'
        no file './radio.so'
        no file '/usr/local/lib/lua/5.1/radio.so'
        no file '/usr/lib/x86_64-linux-gnu/lua/5.1/radio.so'
        no file '/usr/local/lib/lua/5.1/loadall.so'
stack traceback:
        [C]: in function 'error'
        ./radio/core/async.lua:106: in function 'callback'
        ./radio/blocks/sources/hackrf.lua:255: in function 'run'
        ./radio/core/composite.lua:554: in function <./radio/core/composite.lua:554>
        [C]: in function 'xpcall'
        ./radio/core/composite.lua:554: in function 'start'
        ./radio/core/composite.lua:451: in function 'run'
        examples/rtlsdr_ax25.lua:46: in main chunk
        [C]: in function 'dofile'
        ./luaradio:105: in main chunk
        [C]: at 0x004044a0

Diff:
diff --git a/examples/rtlsdr_ax25.lua b/examples/rtlsdr_ax25.lua
index 9d0c4c3..eb97857 100644
--- a/examples/rtlsdr_ax25.lua
+++ b/examples/rtlsdr_ax25.lua
@@ -10,8 +10,10 @@ local tune_offset = -100e3
 local baudrate = 1200

 -- Blocks
-local source = radio.RtlSdrSource(frequency + tune_offset, 1000000)
-local tuner = radio.TunerBlock(tune_offset, 12e3, 80)
+-- local source = radio.RtlSdrSource(frequency + tune_offset, 1000000)
+-- local tuner = radio.TunerBlock(tune_offset, 12e3, 80)
+local source = radio.HackRFSource(frequency + tune_offset, 1000000)
+local tuner  = radio.TunerBlock(tune_offset, 12e3, 80)
 local nbfm_demod = radio.NBFMDemodulator(3e3, 3e3)
 local hilbert = radio.HilbertTransformBlock(129)
 local translator = radio.FrequencyTranslatorBlock(-1700)

dominicgs, 2016-07-09:
I had this issue while experimenting a few nights ago.  It's a path problem.  The code is looking for the radio module that is in the source directory, but the search path doesn't include it.  I'm not sure how the search path is set, but performing `make install` in the `embed` directory fixed the problem for me by installing to the system wide path.

zeryl, 2016-07-09:
That did indeed fix the issue.  I needed to install the libluajit-5.1-dev package on Ubuntu, to satisfy lua.h, and then the embed folder compiled as expected, a make install later, it worked.

vsergeev, 2016-07-09:
@zeryl @dominicgs, ah, yeah, interesting.This is just limited to the new `async.callback()` environment, which doesn't yet inherit the parent's `package.path` (the `luaradio` runner [modifies](https://github.com/vsergeev/luaradio/blob/master/luaradio#L7) it to allow the local imports). It's easy to fix. I'll add the fix so everything can be prototyped out of repo without installs.

vsergeev, 2016-07-09:
Fixed in latest `devel` (1aa8f626d0c83eee41f1fb093d5c9cac7b3723e5). It should be runnable from repo now.

zeryl, 2016-07-09:
Is there any way via the constructor, or similar, to set a PPM value? Perhaps I'm missing it.  I'm trying to decode AX25, and while my error is only -2ppm, I do think it's having a bit of an impact.  That, or AX25 messages are not being decoded at all, but I don't have a good way to test that.

vsergeev, 2016-07-09:
@zeryl, It doesn't seem [libhackrf](https://github.com/mossmann/hackrf/blob/master/host/libhackrf/src/hackrf.h) provides an API for frequency correction, so I have to assume it isn't handled on the device. However, you can compute the correction manually for your frequency of interest, e.g. 144.390 MHz at -2 PPM, `144.390e6 * (1 - 2/1e6) = 144389711.22` and try tuning to 144389711 Hz.

So is the HackRF source working? Could you try it with a simpler demo, like WBFM Stereo, and take a screenshot? I'm curious to see it in action.

zeryl, 2016-07-09:
I don't have audio setup on the box it's running on, but here's a screenshot of the spectrum.

https://i.imgur.com/QI93szo.png

vsergeev, 2016-07-09:
@zeryl, awesome, thanks. That definitely looks like an FM radio station, although the noise floor looks high. As for dropping samples, it's seems unlikely to be a computational issue (it looks like you're running this on a desktop). I just read that the [recommended](https://github.com/mossmann/hackrf/wiki/Tips-and-Tricks#sampling-rate-and-baseband-filters) minimum sample rate is 8 MHz, so maybe you'll have better performance at 8 MHz in the source block, and 32 decimation in the tuner block for WBFM examples, or 640 decimation in the tuner block for the AX25 example. It's also possible there is a bug in the read callback, but that'll get figured out as we make progress.

zeryl, 2016-07-09:
Sounds great, I'm just happy that we have support for HackRF in this! Thank you so much!

vsergeev, 2016-07-13:
@zeryl, no problem -- thanks for testing it out!

@zeryl, Is there any chance you or another (@dominicgs, @rxseger) could give it another try with a supported sample rate (e.g. >= 8 MHz), and confirm there are no dropped samples? I'd feel more comfortable releasing it in v0.2.0 this week if that is the case.

zeryl, 2016-07-13:
What'st he best way to check on dropped samples? I was able to tell on WBFM since the wav file was able to be played.  With 8mhz, just not sure :)

zeryl, 2016-07-13:
@vsergeev Tested with a baseband_bandwidth of 10.0e6, and worked great.  I don't hear any dropped, I don't have a good way to tell, but listened to a WBFM station, and seemed to work great for me.

vsergeev, 2016-07-13:
@zeryl, yeah, for now I just mean empirically -- at the moment there is no easy way to detect dropped samples without some changes to the receive callback or library support for it.

To use a higher sample rate, you'll have to adjust the decimation factor in the TunerBlock, so that the produced WAV file has a reasonable audio rate. But it sounds 10 MHz is running without any obvious issues? That's great to hear.

Thanks for the finding typo, I'll fix it tonight.

commit a3948538 (2016-07-06): blocks/sources/hackrf: add HackRFSource

resolves #18.
commit f76afec6 (2016-07-08): blocks/sources/hackrf: add HackRFSource

resolves #18.
commit d94a7ad8 (2016-07-08): blocks/sources/hackrf: add HackRFSource

resolves #18.
commit fd5868c5 (2016-07-10): blocks/sources/hackrf: add HackRFSource

resolves #18.
closed 2016-07-17 by vsergeev