vsergeev/luaradio #25: Expand on list of dependencies

ul5255, 2016-07-08:
Hi,
Nice project!
On my Scientific Linux 7.2 installation I had to perform the following steps:
* yum install luajit-devel
* yum install fftw3-devel
* yum install gnuplot
* add following line in e.g. ~/.bashrc: export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH
* install liquid-dsp according to web site instructions
* build luaradio
I think the "build from source" section should mention that the "*-devel" packages
for luajit and fftw3 are required.

vsergeev, 2016-07-08:
Cool, thanks for the steps. A couple of comments and questions:

* luajit-devel makes sense, that package provides the pkg-config pc file, as well as the libluajit-5.1.so symlink, on your distribution, both of which are used by the Makefile.
* fftw3-devel -- I'm not exactly sure why you needed this. Did `fftw3f` come up as `false` under `luaradio --platform` without it? Ideally, only the runtime fftw3f library should be needed for fftw3f acceleration.
* gnuplot is needed for the gnuplot plotting sinks -- and in turn to show the plots in the examples --  but is not a hard dependency.
* `LD_LIBRARY_PATH` -- do you remember what this fixed? Which libraries were being installed to /usr/local/lib? If /usr/local/lib is in your ld search paths, then they should get picked up without needing this modification, but it might not be in SL7.

I'll update the installation guide with pre-requisites for the major distributions. Hopefully we can get this process smoothed out soon.

ul5255, 2016-07-08:
If I recall correctly: fftw3-devel was necessary to build liquid-dsp. I dont remember from the top of my head why I had to set LD_LIBRARY_PATH but I can check on Monday. I guess fftw3f and liquid-dsp were shown as disabled when I called luaradio --platform. Regarding gnuplot: I copy/pasted one of your examples which does use a gnuplot sink and it dint run. No big deal to install but I'm thinking more of people who want to test out signal processing and could get discouraged by such error messages. Perhaps you could also have a section which describes the bare minimum install and another section which gives instructions for a full installation. I have a fairly minimal Scientific Linux 7.2 install which could be the reason for those troubles. All of them were fairly easy to fix and the overall installation was just so much easier than GNURadio.

vsergeev, 2016-07-08:
Thanks, and that's fair. I'll try conveying the extra common dependencies in the installation guide.

commit 4df43382 (2016-07-10): docs: expand on prerequisities and dependencies in installation guide

resolves #25.
commit 51af7012 (2016-07-10): docs: expand on prerequisities and dependencies in installation guide

resolves #25.
commit fe792af0 (2016-07-10): docs: expand on prerequisities and dependencies in installation guide

resolves #25.
commit bda4c8d3 (2016-07-10): docs: expand on prerequisities and dependencies in installation guide

resolves #25.
closed 2016-07-10 by vsergeev