vsergeev/luaradio #29: [FrequencyDiscriminatorBlock] Block runtime error: ./radio/core/pipe.lua:199: write(): Connection reset by peer

martinzsmith, 2016-07-17:
This is probably a basic user error, but I do not know enough to debug this myself (yet):
I installed fftw 3.3.4, the latest liquid-dsp, volk, and luajit from source. The hardware is a Odroid-XU4.

```
odroid@odroid:~/src/luaradio$ ./luaradio --platform
luajit          LuaJIT 2.0.4
os              Linux
arch            arm
page size       4096
cpu count       8
cpu model       ARMv7 Processor rev 3 (v7l)
features
    fftw3f      true
    volk        true
    liquid      true

odroid@odroid:~/src/luaradio$ ./luaradio examples/rtlsdr_wbfm_mono.lua 91.1e6
Found Rafael Micro R820T tuner
Exact sample rate is: 1102500.002464 Hz
[FrequencyDiscriminatorBlock] Block runtime error: ./radio/core/pipe.lua:199: write(): Connection reset by peer
stack traceback:
        [C]: in function 'error'
        ./radio/core/pipe.lua:199: in function 'platform_write'
        ./radio/core/pipe.lua:306: in function 'write'
        ./radio/core/block.lua:387: in function 'run'
        ./radio/core/composite.lua:554: in function <./radio/core/composite.lua:554>
        [C]: in function 'xpcall'
        ./radio/core/composite.lua:554: in function 'start'
        ./radio/core/composite.lua:451: in function 'run'
        examples/rtlsdr_wbfm_mono.lua:34: in main chunk
        [C]: in function 'dofile'
        ./luaradio:105: in main chunk
        [C]: at 0x00013721
odroid@odroid:~/src/luaradio$
```

vsergeev, 2016-07-17:
This error means that a downstream block abruptly terminated (crashed). I've only seen it when a gnuplot plotting sink crashes -- which can happen in some setups, see #26.

Are you running this in a graphical environment, or over SSH with X11 forwarding enabled? Could you rerun the example with `-v`?

vsergeev, 2016-07-18:
I've created a separate issue (#30) for improving the error message to be more useful.

martinzsmith, 2016-07-18:
I was using ssh with x11 forwarding to Xming, but I've never had a problem with that before, even gr-fosphor works!

But to eliminate that as the source of the problem I connected a screen and keyboard and ran luaradio directly from the an xterm on the Odroid-XU4, and the below is the output (I ran it 2x, the 2nd time I got one less error message)
```
$ cat console.script.output
$ ./luaradio -v examples/rtlsdr_wbfm_mono.lua 88.5e6
[CompositeBlock] Connected output FrequencyTranslatorBlock.out to input LowpassFilterBlock.in
[CompositeBlock] Connected output LowpassFilterBlock.out to input DownsamplerBlock.in
[CompositeBlock] Aliased input TunerBlock.in to input FrequencyTranslatorBlock.in
[CompositeBlock] Aliased output TunerBlock.out to output DownsamplerBlock.out
[CompositeBlock] Connected output RtlSdrSource.out to input TunerBlock.in
[CompositeBlock] Connected output TunerBlock.out to input FrequencyDiscriminatorBlock.in
[CompositeBlock] Connected output FrequencyDiscriminatorBlock.out to input LowpassFilterBlock.in
[CompositeBlock] Connected output LowpassFilterBlock.out to input FMDeemphasisFilterBlock.in
[CompositeBlock] Connected output FMDeemphasisFilterBlock.out to input DownsamplerBlock.in
[CompositeBlock] Connected output DownsamplerBlock.out to input PulseAudioSink.in
[CompositeBlock] Connected output FrequencyDiscriminatorBlock.out to input GnuplotSpectrumSink.in
[CompositeBlock] Connected output FMDeemphasisFilterBlock.out to input GnuplotSpectrumSink.in
[CompositeBlock] Dependency order:
[CompositeBlock]        RtlSdrSource
[CompositeBlock]            .out   -> {FrequencyTranslatorBlock.in}
[CompositeBlock]        FrequencyTranslatorBlock
[CompositeBlock]            .in    <- {RtlSdrSource.out}
[CompositeBlock]            .out   -> {LowpassFilterBlock.in}
[CompositeBlock]        LowpassFilterBlock
[CompositeBlock]            .in    <- {FrequencyTranslatorBlock.out}
[CompositeBlock]            .out   -> {DownsamplerBlock.in}
[CompositeBlock]        DownsamplerBlock
[CompositeBlock]            .in    <- {LowpassFilterBlock.out}
[CompositeBlock]            .out   -> {FrequencyDiscriminatorBlock.in}
[CompositeBlock]        FrequencyDiscriminatorBlock
[CompositeBlock]            .in    <- {DownsamplerBlock.out}
[CompositeBlock]            .out   -> {LowpassFilterBlock.in, GnuplotSpectrumSink.in}
[CompositeBlock]        LowpassFilterBlock
[CompositeBlock]            .in    <- {FrequencyDiscriminatorBlock.out}
[CompositeBlock]            .out   -> {FMDeemphasisFilterBlock.in}
[CompositeBlock]        FMDeemphasisFilterBlock
[CompositeBlock]            .in    <- {LowpassFilterBlock.out}
[CompositeBlock]            .out   -> {DownsamplerBlock.in, GnuplotSpectrumSink.in}
[CompositeBlock]        GnuplotSpectrumSink
[CompositeBlock]            .in    <- {FrequencyDiscriminatorBlock.out}
[CompositeBlock]        DownsamplerBlock
[CompositeBlock]            .in    <- {FMDeemphasisFilterBlock.out}
[CompositeBlock]            .out   -> {PulseAudioSink.in}
[CompositeBlock]        PulseAudioSink
[CompositeBlock]            .in    <- {DownsamplerBlock.out}
[CompositeBlock]        GnuplotSpectrumSink
[CompositeBlock]            .in    <- {FMDeemphasisFilterBlock.out}
[CompositeBlock] Parent pid 26368
[CompositeBlock] Block RtlSdrSource pid 26373
[CompositeBlock] Block FrequencyTranslatorBlock pid 26374
[CompositeBlock] Block LowpassFilterBlock pid 26375
[CompositeBlock] Block FrequencyDiscriminatorBlock pid 26378
[CompositeBlock] Block DownsamplerBlock pid 26376
[CompositeBlock] Block LowpassFilterBlock pid 26379
[CompositeBlock] Block FMDeemphasisFilterBlock pid 26380
[CompositeBlock] Block GnuplotSpectrumSink pid 26381
[CompositeBlock] Block DownsamplerBlock pid 26382
[CompositeBlock] Block PulseAudioSink pid 26383
[CompositeBlock] Block GnuplotSpectrumSink pid 26384
Found Rafael Micro R820T tuner
[RtlSdrSource] Frequency: 88250000 Hz, Sample rate: 1102500 Hz
Exact sample rate is: 1102500.002464 Hz
[FrequencyDiscriminatorBlock] Block runtime error: ./radio/core/pipe.lua:199: write(): Connection reset by peer
stack traceback:
        [C]: in function 'error'
        ./radio/core/pipe.lua:199: in function 'platform_write'
        ./radio/core/pipe.lua:306: in function 'write'
        ./radio/core/block.lua:387: in function 'run'
        ./radio/core/composite.lua:554: in function <./radio/core/composite.lua:554>
[... 72 lines cut ...]
[CompositeBlock] Block RtlSdrSource pid 26396
[CompositeBlock] Block FrequencyDiscriminatorBlock pid 26400
[CompositeBlock] Block DownsamplerBlock pid 26404
[CompositeBlock] Block FMDeemphasisFilterBlock pid 26403
[CompositeBlock] Block PulseAudioSink pid 26407
Found Rafael Micro R820T tuner
[RtlSdrSource] Frequency: 88250000 Hz, Sample rate: 1102500 Hz
Exact sample rate is: 1102500.002464 Hz
[CompositeBlock] Child exited. Shutting down...
[FrequencyDiscriminatorBlock] Block runtime error: ./radio/core/pipe.lua:199: write(): Connection reset by peer
stack traceback:
        [C]: in function 'error'
        ./radio/core/pipe.lua:199: in function 'platform_write'
        ./radio/core/pipe.lua:306: in function 'write'
        ./radio/core/block.lua:387: in function 'run'
        ./radio/core/composite.lua:554: in function <./radio/core/composite.lua:554>
        [C]: in function 'xpcall'
        ./radio/core/composite.lua:554: in function 'start'
        ./radio/core/composite.lua:451: in function 'run'
        examples/rtlsdr_wbfm_mono.lua:34: in main chunk
        [C]: in function 'dofile'
        ./luaradio:105: in main chunk
        [C]: at 0x0000bfb9
$ exit
```

vsergeev, 2016-07-18:
Yeah, I think the gnuplot sinks are crashing. They can be a bit finicky, it seems. In my case, I've seen them crash if I have a mouse cursor on the plot window before they've started plotting, but otherwise they work fairly well for me. I'll have to investigate further and see if there is a underlying bug common to all of these immediate gnuplot crashes.

In the short term, I would give the `wxt` terminal a try, which is what I've been using (see this [comment](https://github.com/vsergeev/luaradio/issues/26#issuecomment-232576790) for the configuration line).

Alternatively, you can also disable plotting for the examples by unsetting the `DISPLAY` environment variable (but that's not useful if it's the plots you're after).

In general, I think LuaRadio could use plotting sinks that use a more stable backend  (I don't think gnuplot isn't really meant for this kind of real-time dynamic plotting, and it's a wonder they work at all). I've put this on the [project roadmap](https://github.com/vsergeev/luaradio/wiki#project-roadmap).

martinzsmith, 2016-07-18:
I'm upgrading the OS at the moment from:
Ubuntu 15.04 (Vivid Vervet)
to:
Ubuntu 15.10 (Wily Werewolf)

In case it is an older OS problem, I'll give wxt a spin after the upgrade is finished.

vsergeev, 2016-07-18:
Sure thing. I'm running `gnuplot 5.0 patchlevel 3` (on Arch Linux).

martinzsmith, 2016-07-19:
Still no joy.
$ gnuplot -V
gnuplot 4.6 patchlevel 6
$ sudo apt-get remove gnuplot
$ sudo apt-get install gnuplot5
$ gnuplot -V
gnuplot 5.0 patchlevel 1

vsergeev, 2016-07-20:
Could you try running it with plotting disabled, to confirm it's the plotting sinks that are crashing?

```
$ unset DISPLAY
$ luaradio examples/rtlsdr_wbfm_mono.lua 91.1e6
```

martinzsmith, 2016-07-21:
I tried that and I still get the same error. I'm just going to do a clean OS install of 16.04 right now and I'll let you know.

martinzsmith, 2016-07-21:
I did a 100% clean install of Ubuntu Mate 16.04, installed almost everything using apt-get (except for liquid-dsp).
It works perfectly on the console:
http://i.imgur.com/RP1jfXI.png
It hangs when I am stupid and try to run it over X which makes total sense, since there is no way to pipe the audio over X and the Odroid is logged out from X windows so there is no way to play back the audio.  but with "unset DISPLAY" it generates the wbfm_mono.wav file as expected. 

I have not traced back the root cause of the problem, but there is no harm ever in a fresh clean OS install.

closed 2016-07-21 by martinzsmith

martinzsmith, 2016-07-21:
It happened again! But this time I found the cause and a workaround.
I always end up running "volk_profile", eventually, to optimise the DSP selection on my hardware. For some reason on ARM luaradio does not like this. My quickfix was "rm ~/.volk/volk_config" and suddenly luaradio was back working again.

vsergeev, 2016-07-22:
Interesting. @martinzsmith, could you try running the example with VOLK disabled:

```
$ LUARADIO_DISABLE_VOLK=1 luaradio examples/rtlsdr_wbfm_mono.lua 91.1e6
```

and see if the crash comes up again? This way we can isolate it to VOLK acceleration, and perhaps next we can narrow down the crashing block. If we can find a repeatable setup, we could determine if it's an issue with VOLK or LuaRadio interfacing with it.

Also, note that VOLK is at 1.3, so if there is an issue there, it may have been fixed recently.

As for X11 forwarding and audio, the PulseAudio sink should still try to play audio locally (it won't forward over SSH), but forwarding the real-time plots will probably be prohibitively slow over SSH, and slow the flow graph to a crawl...

martinzsmith, 2016-07-25:
That works I just noticed that it was a different warning message:
(with LUARADIO_DISABLE_VOLK=1 it works correctly)

```
odroid@odroid:~/src/luaradio$ LUARADIO_DISABLE_VOLK=0 ./luaradio -v examples/rtlsdr_wbfm_mono.lua 88.5e6
[CompositeBlock] Connected output FrequencyTranslatorBlock.out to input LowpassFilterBlock.in
[CompositeBlock] Connected output LowpassFilterBlock.out to input DownsamplerBlock.in
[CompositeBlock] Aliased input TunerBlock.in to input FrequencyTranslatorBlock.in
[CompositeBlock] Aliased output TunerBlock.out to output DownsamplerBlock.out
[CompositeBlock] Connected output RtlSdrSource.out to input TunerBlock.in
[CompositeBlock] Connected output TunerBlock.out to input FrequencyDiscriminatorBlock.in
[CompositeBlock] Connected output FrequencyDiscriminatorBlock.out to input LowpassFilterBlock.in
[CompositeBlock] Connected output LowpassFilterBlock.out to input FMDeemphasisFilterBlock.in
[CompositeBlock] Connected output FMDeemphasisFilterBlock.out to input DownsamplerBlock.in
[CompositeBlock] Connected output DownsamplerBlock.out to input PulseAudioSink.in
[CompositeBlock] Connected output FrequencyDiscriminatorBlock.out to input GnuplotSpectrumSink.in
[CompositeBlock] Connected output FMDeemphasisFilterBlock.out to input GnuplotSpectrumSink.in
[CompositeBlock] Dependency order:
[CompositeBlock]        RtlSdrSource
[CompositeBlock]            .out   -> {FrequencyTranslatorBlock.in}
[CompositeBlock]        FrequencyTranslatorBlock
[CompositeBlock]            .in    <- {RtlSdrSource.out}
[CompositeBlock]            .out   -> {LowpassFilterBlock.in}
[CompositeBlock]        LowpassFilterBlock
[CompositeBlock]            .in    <- {FrequencyTranslatorBlock.out}
[CompositeBlock]            .out   -> {DownsamplerBlock.in}
[CompositeBlock]        DownsamplerBlock
[CompositeBlock]            .in    <- {LowpassFilterBlock.out}
[CompositeBlock]            .out   -> {FrequencyDiscriminatorBlock.in}
[CompositeBlock]        FrequencyDiscriminatorBlock
[CompositeBlock]            .in    <- {DownsamplerBlock.out}
[CompositeBlock]            .out   -> {LowpassFilterBlock.in, GnuplotSpectrumSink.in}
[CompositeBlock]        GnuplotSpectrumSink
[CompositeBlock]            .in    <- {FrequencyDiscriminatorBlock.out}
[CompositeBlock]        LowpassFilterBlock
[CompositeBlock]            .in    <- {FrequencyDiscriminatorBlock.out}
[CompositeBlock]            .out   -> {FMDeemphasisFilterBlock.in}
[CompositeBlock]        FMDeemphasisFilterBlock
[CompositeBlock]            .in    <- {LowpassFilterBlock.out}
[CompositeBlock]            .out   -> {DownsamplerBlock.in, GnuplotSpectrumSink.in}
[CompositeBlock]        DownsamplerBlock
[CompositeBlock]            .in    <- {FMDeemphasisFilterBlock.out}
[CompositeBlock]            .out   -> {PulseAudioSink.in}
[CompositeBlock]        PulseAudioSink
[CompositeBlock]            .in    <- {DownsamplerBlock.out}
[CompositeBlock]        GnuplotSpectrumSink
[CompositeBlock]            .in    <- {FMDeemphasisFilterBlock.out}
[CompositeBlock] Parent pid 4738
[CompositeBlock] Block RtlSdrSource pid 4753
[CompositeBlock] Block FrequencyTranslatorBlock pid 4754
[CompositeBlock] Block LowpassFilterBlock pid 4755
[CompositeBlock] Block GnuplotSpectrumSink pid 4759
[CompositeBlock] Block FrequencyDiscriminatorBlock pid 4758
[CompositeBlock] Block FMDeemphasisFilterBlock pid 4761
[CompositeBlock] Block PulseAudioSink pid 4763
[CompositeBlock] Block GnuplotSpectrumSink pid 4764
[CompositeBlock] Block DownsamplerBlock pid 4756
[CompositeBlock] Block LowpassFilterBlock pid 4760
[CompositeBlock] Block DownsamplerBlock pid 4762
Found Rafael Micro R820T tuner
[RtlSdrSource] Frequency: 88250000 Hz, Sample rate: 1102500 Hz
Exact sample rate is: 1102500.002464 Hz
Using Volk machine: neon_hardfp_orc
Using Volk machine: neon_hardfp_orc
Volk warning: no arch found, returning generic impl
Using Volk machine: neon_hardfp_orc
Using Volk machine: neon_hardfp_orc
[CompositeBlock] Child exited. Shutting down...
Using Volk machine: neon_hardfp_orc
odroid@odroid:~/src/luaradio$ LUARADIO_DISABLE_VOLK=1 ./luaradio examples/rtlsdr_wbfm_mono.lua 88.5e6
Found Rafael Micro R820T tuner
Exact sample rate is: 1102500.002464 Hz

odroid@odroid:~/src/luaradio$
```

martinzsmith, 2016-07-25:
The volk_config file is more sparse than I would have expected it to be. So it looks like this is a new volk problem and not a problem with luaradio. I'll remove the package and install the latest volk from source and let you know:
```
odroid@odroid:~/src/luaradio$ rm ~/.volk/volk_config
odroid@odroid:~/src/luaradio$ volk_profile
Using Volk machine: neon_hardfp_orc
RUN_VOLK_TESTS: volk_64u_popcntpuppet_64u(131071,1987)
generic completed in 647.898ms
neon completed in 1698.83ms
Best aligned arch: generic
Best unaligned arch: generic
RUN_VOLK_TESTS: volk_16u_byteswappuppet_16u(131071,1987)
generic completed in 560.498ms
neon completed in 241.432ms
neon_table completed in 221.543ms
Best aligned arch: neon_table
Best unaligned arch: neon_table
RUN_VOLK_TESTS: volk_32u_byteswappuppet_32u(131071,1987)
generic completed in 519.355ms
neon completed in 456.014ms
Best aligned arch: neon
Best unaligned arch: neon
RUN_VOLK_TESTS: volk_32u_popcntpuppet_32u(131071,1987)
no architectures to test
RUN_VOLK_TESTS: volk_64u_byteswappuppet_64u(131071,1987)
generic completed in 948.137ms
neon completed in 1084.04ms
Best aligned arch: generic
Best unaligned arch: generic
RUN_VOLK_TESTS: volk_32fc_s32fc_rotatorpuppet_32fc(131071,1987)
no architectures to test
RUN_VOLK_TESTS: volk_8u_conv_k7_r2puppet_8u(131071,198)
no architectures to test
RUN_VOLK_TESTS: volk_32f_x2_fm_detectpuppet_32f(131071,1987)
no architectures to test
RUN_VOLK_TESTS: volk_16ic_s32f_deinterleave_real_32f(131071,1987)
no architectures to test
RUN_VOLK_TESTS: volk_16ic_deinterleave_real_8i(131071,1987)
generic completed in 358.775ms
neon completed in 116.201ms
u_orc completed in 181.331ms
Best aligned arch: neon
Best unaligned arch: neon
RUN_VOLK_TESTS: volk_16ic_deinterleave_16i_x2(131071,1987)
generic completed in 510.308ms
u_orc completed in 230.002ms
Best aligned arch: u_orc
Best unaligned arch: u_orc
RUN_VOLK_TESTS: volk_16ic_s32f_deinterleave_32f_x2(131071,1987)
generic completed in 1288.82ms
neon completed in 676.882ms
u_orc completed in 1589.96ms
Best aligned arch: neon
Best unaligned arch: neon
RUN_VOLK_TESTS: volk_16ic_deinterleave_real_16i(131071,1987)
no architectures to test
RUN_VOLK_TESTS: volk_16ic_magnitude_16i(131071,1987)
no architectures to test
RUN_VOLK_TESTS: volk_16ic_s32f_magnitude_32f(131071,1987)
no architectures to test
RUN_VOLK_TESTS: volk_16i_s32f_convert_32f(131071,1987)
generic completed in 624.154ms
neon completed in 270.31ms
a_generic completed in 606.278ms
Best aligned arch: neon
Best unaligned arch: neon
RUN_VOLK_TESTS: volk_16i_convert_8i(131071,1987)
generic completed in 437.672ms
neon completed in 56.42ms
a_generic completed in 428.233ms
Best aligned arch: neon
[... 421 lines cut ...]
volk_32f_x2_multiply_32f neon neon
volk_32f_s32f_normalize u_orc u_orc

volk_32f_sqrt_32f neon neon

volk_32f_x2_subtract_32f u_orc u_orc
volk_32f_x3_sum_of_poly_32f neonvert neonvert
volk_32i_x2_and_32i neon neon
volk_32i_s32f_convert_32f a_generic generic
volk_32i_x2_or_32i u_orc u_orc

volk_64f_convert_32f generic generic

volk_8ic_deinterleave_real_8i neon neon

volk_8i_convert_16i neon neon
volk_8i_s32f_convert_32f a_generic generic
volk_32fc_s32fc_multiply_32fc neon neon
volk_32f_s32f_multiply_32f u_orc u_orc
volk_32f_binary_slicer_32i generic_branchless generic_branchless
volk_32f_binary_slicer_8i neon neon
volk_32f_tanh_32f series series

odroid@odroid:~/src/luaradio$
```

vsergeev, 2016-07-27:
@martinzsmith, thanks for looking into it. It does seem that VOLK is crashing somewhere internally. It could technically still be LuaRadio's fault, if, for example, it passes unaligned data to a VOLK function expecting aligned data, but as far as I know this shouldn't be happening.

Let me know if VOLK 1.3 fixes the issue.