vsergeev/luaradio #36: Not compatible with Macports

gedger, 2016-08-02:
Currently luaradio will not find libraries or applications that have been installed using Macports as they are typically installed in /opt/... Even if applications are on the path they are not found e.g. gnuplot.

A good solution would be to add luaradio to Macports and I have drafted a Portfile which currently builds and installs luaradio. However, it raises warnings as the luaradio Makefile tries to install libs
into /usr/local. I think it's something to do with the Makefile not handling a passed PREFIX and/or DESTDIR.

I have limited experience of writing Portfiles and Makefiles but happy to help get luaradio into Macports if that's something you would like to do.

rht, 2016-10-16:
This is a possible duplicate with #13.

vsergeev, 2020-03-23:
Package added to MacPorts with macports/macports-ports#6539 with a wrapper script to fix dynamic library loading in LuaRadio.

closed 2020-03-23 by vsergeev