vsergeev/luaradio #39: Raspberry Pi 2 luaradio example not running

Godtec, 2016-12-18:
Hi there:

After installing luaradio twice and re-compiling liquid-Dsp twice.
I cannot seem to get the luaradio example to run. Below is the error I am running into.
Any help is very much appreciated!! 

-------------------------------
mikek@raspberrypi:~/luaradio $ ./luaradio examples/rtlsdr_ax25.lua 
Usage: examples/rtlsdr_ax25.lua <frequency>
mikek@raspberrypi:~/luaradio $ ./luaradio examples/rtlsdr_ax25.lua 144.390e6
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
Using Volk machine: neon_hardfp_orc
Using Volk machine: neon_hardfp_orc
[FrequencyTranslatorBlock] Downstream block LowpassFilterBlock terminated unexpectedly.
[FrequencyTranslatorBlock] Block runtime error: ./radio/core/pipe.lua:300: write(): Broken pipe
stack traceback:
	[C]: in function 'error'
	./radio/core/pipe.lua:300: in function 'write'
	./radio/core/block.lua:387: in function 'run'
	./radio/core/composite.lua:558: in function <./radio/core/composite.lua:558>
	[C]: in function 'xpcall'
	./radio/core/composite.lua:558: in function 'start'
	./radio/core/composite.lua:452: in function 'run'
	examples/rtlsdr_ax25.lua:44: in main chunk
	[C]: in function 'dofile'
	./luaradio:105: in main chunk
	[C]: at 0x0000c790
[RtlSdrSource] Downstream block FrequencyTranslatorBlock terminated unexpectedly.
[RtlSdrSource] Block runtime error: ./radio/core/pipe.lua:300: write(): Broken pipe
stack traceback:
	[C]: in function 'error'
	./radio/core/pipe.lua:300: in function 'write'
	./radio/blocks/sources/rtlsdr.lua:210: in function <./radio/blocks/sources/rtlsdr.lua:198>
	[C]: in function 'rtlsdr_read_async'
	./radio/blocks/sources/rtlsdr.lua:228: in function 'run'
	./radio/core/composite.lua:558: in function <./radio/core/composite.lua:558>
	[C]: in function 'xpcall'
	./radio/core/composite.lua:558: in function 'start'
	./radio/core/composite.lua:452: in function 'run'
	examples/rtlsdr_ax25.lua:44: in main chunk
	[C]: in function 'dofile'
	./luaradio:105: in main chunk
	[C]: at 0x0000c790
mikek@raspberrypi:~/luaradio $ 
------------------------------------

MikeK

vsergeev, 2017-01-06:
Hi Mike,

Are you running this in a graphical environment? It looks like a plotting sink might be crashing.

In general, the RPi2 can't keep up with the signal processing at the 1 MHz sample rate used in many of the examples. They will run on the RPi3, which is much more performant than the RPi2.

You could *try* modifying this example to run at one lower sample rates in the 200 - 300 kHz range, as this example really doesn't need the 1 MHz bandwidth. For example, at 250 kHz:

```
...
local tune_offset = -50e3
local baudrate = 1200

-- Blocks
local source = radio.RtlSdrSource(frequency + tune_offset, 250000)
local tuner = radio.TunerBlock(tune_offset, 12e3, 20)
...
```

vsergeev, 2019-12-13:
Closing for now. Feel free to reopen if the issue resurfaces.

closed 2019-12-13 by vsergeev