vsergeev/luaradio #54: [Install] Update homebrew package with v0.6.0

vsergeev, 2019-12-20:
I currently don't have access to a Mac for development.

umi-neko, 2019-12-26:
Hi,
Glad to see luaradio release new version.
I am using brew install luaradio, but it still try to install v0.5.1 for me. Do I need to add some tap for v0.6.0?

vsergeev, 2019-12-26:
LuaRadio is in homebrew/core, but the package just needs to be updated for v0.6.0. I just don't currently have access to a Mac / VM to make the update, test it, and submit a PR to homebrew.

BG7JAF, 2019-12-27:
Hello,
I clone the git source and try to run the FM example code on macOS 10.15.2, but it can't work. Seems it's incompatible with pulseaudio which version is 13.0 (installed from brew).
Here is the error messages:

➜  luaradio git:(master) ✗ ./luaradio fm.lua
Found Rafael Micro R820T tuner
Exact sample rate is: 1102500.002464 Hz
W: [] caps.c: Normally all extra capabilities would be dropped now, but that's impossible because PulseAudio was built without capabilities support.
[PulseAudioSink] Block runtime error: ./radio/blocks/sinks/pulseaudio.lua:108: pa_simple_new(): Connection refused
stack traceback:
	[C]: in function 'error'
	./radio/blocks/sinks/pulseaudio.lua:108: in function 'initialize_pulseaudio'
	./radio/blocks/sinks/pulseaudio.lua:119: in function 'process'
	./radio/core/block.lua:399: in function 'run'
	./radio/core/composite.lua:539: in function <./radio/core/composite.lua:539>
	[C]: in function 'xpcall'
	./radio/core/composite.lua:539: in function 'start'
	./radio/core/composite.lua:432: in function 'run'
	fm.lua:11: in main chunk
	[C]: in function 'dofile'
	./luaradio:110: in main chunk
	[C]: at 0x010b86c300
[DownsamplerBlock] Downstream block PulseAudioSink terminated unexpectedly.
[DownsamplerBlock] Block runtime error: ./radio/core/pipe.lua:310: write(): Broken pipe
stack traceback:
	[C]: in function 'error'
	./radio/core/pipe.lua:310: in function 'write'
	./radio/core/block.lua:404: in function 'run'
	./radio/core/composite.lua:539: in function <./radio/core/composite.lua:539>
	[C]: in function 'xpcall'
	./radio/core/composite.lua:539: in function 'start'
	./radio/core/composite.lua:432: in function 'run'
	fm.lua:11: in main chunk
	[C]: in function 'dofile'
	./luaradio:110: in main chunk
	[C]: at 0x010b86c300
[FMDeemphasisFilterBlock] Downstream block DownsamplerBlock terminated unexpectedly.
[FMDeemphasisFilterBlock] Block runtime error: ./radio/core/pipe.lua:310: write(): Broken pipe
stack traceback:
	[C]: in function 'error'
	./radio/core/pipe.lua:310: in function 'write'
	./radio/core/block.lua:404: in function 'run'
	./radio/core/composite.lua:539: in function <./radio/core/composite.lua:539>
	[C]: in function 'xpcall'
	./radio/core/composite.lua:539: in function 'start'
	./radio/core/composite.lua:432: in function 'run'
	fm.lua:11: in main chunk
	[C]: in function 'dofile'
	./luaradio:110: in main chunk
	[C]: at 0x010b86c300
[LowpassFilterBlock] Downstream block FMDeemphasisFilterBlock terminated unexpectedly.
[LowpassFilterBlock] Block runtime error: ./radio/core/pipe.lua:310: write(): Broken pipe
stack traceback:
	[C]: in function 'error'
	./radio/core/pipe.lua:310: in function 'write'
	./radio/core/block.lua:404: in function 'run'
	./radio/core/composite.lua:539: in function <./radio/core/composite.lua:539>
	[C]: in function 'xpcall'
	./radio/core/composite.lua:539: in function 'start'
	./radio/core/composite.lua:432: in function 'run'
	fm.lua:11: in main chunk
	[C]: in function 'dofile'
	./luaradio:110: in main chunk
	[C]: at 0x010b86c300
[FrequencyDiscriminatorBlock] Downstream block LowpassFilterBlock terminated unexpectedly.
[FrequencyDiscriminatorBlock] Block runtime error: ./radio/core/pipe.lua:310: write(): Broken pipe
stack traceback:
	[C]: in function 'error'
	./radio/core/pipe.lua:310: in function 'write'
	./radio/core/block.lua:404: in function 'run'
[... 41 lines cut ...]
	./radio/core/block.lua:404: in function 'run'
	./radio/core/composite.lua:539: in function <./radio/core/composite.lua:539>
	[C]: in function 'xpcall'
	./radio/core/composite.lua:539: in function 'start'
	./radio/core/composite.lua:432: in function 'run'
	fm.lua:11: in main chunk
	[C]: in function 'dofile'
	./luaradio:110: in main chunk
	[C]: at 0x010b86c300
[RtlSdrSource] Downstream block FrequencyTranslatorBlock terminated unexpectedly.
[RtlSdrSource] Block runtime error: ./radio/core/pipe.lua:310: write(): Broken pipe
stack traceback:
	[C]: in function 'error'
	./radio/core/pipe.lua:310: in function 'write'
	./radio/blocks/sources/rtlsdr.lua:231: in function <./radio/blocks/sources/rtlsdr.lua:219>
	[C]: in function 'rtlsdr_read_async'
	./radio/blocks/sources/rtlsdr.lua:249: in function 'run'
	./radio/core/composite.lua:539: in function <./radio/core/composite.lua:539>
	[C]: in function 'xpcall'
	./radio/core/composite.lua:539: in function 'start'
	./radio/core/composite.lua:432: in function 'run'
	fm.lua:11: in main chunk
	[C]: in function 'dofile'
	./luaradio:110: in main chunk
	[C]: at 0x010b86c300

vsergeev, 2019-12-27:
@BG7JAF, is the PulseAudio server running? You may need to start it with `brew services start pulseaudio`.

umi-neko, 2019-12-27:
@vsergeev it's the reason!  After follow your instruction start the pulseaudio service, it can work now. I will do other test. Thankyou

vsergeev, 2020-02-05:
Resolved with homebrew/homebrew-core#49807

closed 2020-02-05 by vsergeev