vsergeev/luaradio #56: PulseAudio initialize does'nt work

reario, 2019-12-28:
HI,
I've successful compiled and installed LuaRadio.
I'm now experienced problems with PulseRadio sink: everytime I use it I always get the error:
"[PulseAudioSink] Block runtime error: ./radio/blocks/sinks/pulseaudio.lua:108: pa_simple_new(): Invalid argument"
Simple program is like this:
------------------------------------
ocal radio = require('radio')

-- Blocks
local source1 = radio.SignalSource('sine', 10e3, 1e6) -- 10 kHz cosine source, sampled at 1 MHz
local throttle = radio.ThrottleBlock()
--local sink = radio.PulseAudioSink(1)
local sink = os.getenv('DISPLAY') and radio.PulseAudioSink(1) or radio.WAVFileSink('wbfm_mono.wav', 1)
-- local sink = radio.PortAudioSink(1)
local gnuplot = radio.GnuplotPlotSink(1000,'Sine')
top = radio.CompositeBlock()
--top:connect(source1,throttle,gnuplot)
top:connect(source1,sink)
-- Run it
top:run()
-----------------------------
Is it an error or my installation fault?
thanks, vittorio

reario, 2019-12-29:
solved. I was sampling at 1e6 sample per second......
It was enogh to reduce it at 20KHz and all worked

vsergeev, 2020-01-07:
Thanks for the follow up. It sounds like the sample rate was too high for PulseAudio. Unfortunately, the error reporting the PulseAudio Simple API is somewhat limited. Closing issue for now.

closed 2020-01-07 by vsergeev