vsergeev/luaradio #59: Unable to get luaradio to run from installed location but runs from checkout

phrrngtn, 2020-02-05:
I built luaradio from source (cd embed; make lib). I am not at all familiar with lua and the layout of installations and packages so there may be a very simple answer to my issue. My end goal is to be able to run luaradio scripts from Python using the C luaradio library.

The luaradio script from when executed with CWD set to the git checkout; it does not run from any other location. I have not been able to find files that I would expect to be part of the installation (e.g. ax25framer.lua) underneath /usr nor /usr/local. 

```
phrrngtn@sdrpi:~/work $ /usr/local/bin/luaradio --platform
Segmentation fault

phrrngtn@sdrpi:~/work $ cd luaradio/
phrrngtn@sdrpi:~/work/luaradio $ /usr/local/bin/luaradio --platform 
luajit          LuaJIT 2.1.0-beta3
os              Linux
arch            arm
page size       4096
cpu count       4
cpu model       ARMv7 Processor rev 3 (v7l)
features
    fftw3f      true    fftw-3.3.8-neon
    volk        true    2.1 (neonv7_hardfp)
    liquid      true    1.3.2
```

I was able to use the library (minimally: I have not tried to run any scripts yet) from Python after adding -lluajit-5.1 to the LD_FLAGS in the Makefile. Prior to that, any attempt to load the library resulted in a stacktrace with an undefined symblo error for lua_gettop 

```
phrrngtn@sdrpi:~ $ python3
Python 3.7.3 (default, Apr  3 2019, 05:39:12)
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ctypes
>>> luaradio= ctypes.CDLL('libluaradio.so')
>>> luaradio.luaradio_version.restype = ctypes.c_char_p
>>> luaradio.luaradio_version()
b'v0.6.0'

phrrngtn@sdrpi:~ $ ldd /usr/local/lib/libluaradio.so
        linux-vdso.so.1 (0xbeef6000)
        /usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0xb6eb4000)
        libluajit-5.1.so.2 => /usr/lib/arm-linux-gnueabihf/libluajit-5.1.so.2 (0xb6e26000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb6cd8000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb6c56000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb6c43000)
        libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb6c16000)
        /lib/ld-linux-armhf.so.3 (0xb6f36000)

```

vsergeev, 2020-02-08:
I wasn't able to reproduce the segfault on an RPi3 with LuaJIT 2.0.5 or LuaJIT 2.1.0-beta3. Are you on an RPi4? and which OS?

phrrngtn, 2020-02-08:
It is a Raspberry Pi 4. Note that I manually changed the Makefile so that it linked against libluajit-5.1.so.2 .... the Makefile in embed does not link against luajit so my ctypes.CDLL('libluaradio.so') was failing in Python.  If I preloaded (via ctypes, not LD_PRELOAD!) libluajit-5.1, then loading of the standard libluaradio.so succeeded (presumably as all the various dependencies were satisfied).

This is probably a trivial problem (or more likely 2 trivial problems: one with the linking and one with the lua path) and there is probably no need for you to worry about it. I am confused by the numbering of the apt packages, Lua language revisions, luajit revisions etc. About 95% of my C development was done in SunOS and Solaris so now I have to learn  21'st century toolchains, ARM Linux, Lua and SDR all in one go.  
```
phrrngtn@sdrpi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

phrrngtn@sdrpi:~ $ uname -a
Linux sdrpi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
```

vsergeev, 2020-02-11:
I had tried it with and without the extra linkage to libluajit for libluaradio on the RPi3, but no luck on the segfault. I'm going to give a try tomorrow on the RPi4 and I'll see if I can reproduce it there.

As for the linking problem, it's a little complicated because libluaradio is used as a dual library for both standalone applications via the libluaradio C API, and as a Lua module (via the `/usr/local/lib/lua/5.1/radio.so` symlink) that can be loaded with `require()`. For the Lua module case (radio.so), the preferred distribution is without linking to libluajit. This way applications can provide their selected LuaJIT core and version, either dynamically or statically linked, and can load the Lua module without pulling in a potentially different LuaJIT core. On the other hand, for the standalone embedding case (libluaradio.so), it's a little inconvenient to have to manually supply the Lua core in addition to libluaradio.

If it's not too bad to load libluajit with ctypes first in Python, I would stick to that approach for now. If this proves to be too much of a nuisance among users down the line, the Makefile could build and install two versions: libluaradio linked with libluajit, and the radio.so module without the link.

vsergeev, 2020-02-12:
I tried it on the RPi4 without luck reproducing the segfault, although I'm running a different Linux (Arch Linux ARM). Could you try running it in a safe mode with `LUARADIO_DISABLE_LIQUID=1 LUARADIO_DISABLE_VOLK=1 LUARADIO_DISABLE_FFTW3F=1 luaradio --platform`, in case one of the acceleration libraries is the culprit? Also, does the segfault happen without the LDFLAGS modification?

phrrngtn, 2020-02-13:
I reset to HEAD and rebuilt and run a make install and then run as you 
suggested (with the various env vars set) and tt did not make any 
difference. However, I think that the problem is very likely a simple lua 
packaging one. I do not see anywhere in the Makefile where the lua source 
files for 'radio' are being installed. Is there something going on that 
bundles them into the .so?  I'm afraid I don't understand these two lines 
of code at all but that is certainly the first require is certainly the 
location of the segfault.

-- Preload the radio package
-- Possible package paths:
--  Local package:      e.g. ./radio/init.lua
--  Lua package:        e.g. /usr/share/lua/5.1/radio/init.lua
--  Local C package:    e.g. ./radio.so
--  C package:          e.g. /usr/lib/lua/5.1/radio.so
package.preload['radio'] = require('radio')

local radio = require('radio')

pjjH

phrrngtn@sdrpi:~ $ LUARADIO_DISABLE_LIQUID=1 
LUARADIO_DISABLE_VOLK=1 LUARADIO_DISABLE_FFTW3F=1 luaradio --platform
Segmentation fault

All the tests run by busted pass *except* for these two:

Failure  ./tests/jigs.lua @ 84
FIRFilterBlock test vector 1 Dot product, 1 Float32 tap, 256 
ComplexFloat32 input, 256 ComplexFloat32 output
./tests/jigs.lua:20: Expected objects to be the same.
Passed in:
(boolean) false
Expected:
(boolean) true

Failure  ./tests/jigs.lua @ 84
FIRFilterBlock test vector 9 Dot product, 1 Float32 tap, 256 Float32 
input, 256 Float32 output
./tests/jigs.lua:20: Expected objects to be the same.
Passed in:
(boolean) false
Expected:
(boolean) true

vsergeev, 2020-02-15:
Yup, the LuaRadio modules are bytecode compiled and bundled into the .so. Would you mind confirming if this causes a segfault outside of the git repo root, with a vanilla install (no libluajit linking): `luajit -e "require('radio')"`

vsergeev, 2020-04-02:
Closing for now. Let me know if you continue to encounter the out-of-tree segfault. Attached is an example of launching a LuaRadio flow graph and and interacting with it from Python:

``` py
import ctypes
import os
import struct

# Load libluajit with globally exported symbols, so the Lua 5.1 C API can be
# resolved by libluaradio
libluajit = ctypes.CDLL('libluajit-5.1.so', mode=ctypes.RTLD_GLOBAL)

# Load libluaradio
libluaradio = ctypes.CDLL('./embed/build/libluaradio.so')

# Declare API prototypes
libluaradio.luaradio_version.argtypes = []
libluaradio.luaradio_version.restype = ctypes.c_char_p

libluaradio.luaradio_new.argtypes = []
libluaradio.luaradio_new.restype = ctypes.c_void_p

libluaradio.luaradio_load.argtypes = [ctypes.c_void_p, ctypes.c_char_p]
libluaradio.luaradio_load.restype = ctypes.c_int

libluaradio.luaradio_start.argtypes = [ctypes.c_void_p]
libluaradio.luaradio_start.restype = ctypes.c_int

libluaradio.luaradio_stop.argtypes = [ctypes.c_void_p]
libluaradio.luaradio_stop.restype = ctypes.c_int

libluaradio.luaradio_wait.argtypes = [ctypes.c_void_p]
libluaradio.luaradio_wait.restype = ctypes.c_int

libluaradio.luaradio_free.argtypes = [ctypes.c_void_p]
libluaradio.luaradio_free.restype = None

script = """
    return radio.CompositeBlock():connect(
        radio.SignalSource('constant', 0, 1e6, {{amplitude=1.5}}),
        radio.MultiplyConstantBlock(2.0),
        radio.RealFileSink({:d}, 'f32le')
    )
"""

print(libluaradio.luaradio_version())

radio = libluaradio.luaradio_new()

# Create a pipe to share data between the LuaRadio flow graph and Python
snk_rd_fd, snk_wr_fd = os.pipe()

# Hand off the write fd of the pipe to the LuaRadio flow graph
assert(libluaradio.luaradio_load(radio, script.format(snk_wr_fd).encode()) == 0)

# Start LuaRadio flow graph
assert(libluaradio.luaradio_start(radio) == 0)

while True:
    # Read a sample from the read fd of the pipe
    data = os.read(snk_rd_fd, 4)
    # Unpack it as a float
    sample = struct.unpack('f', data)
    # Print the sample
    print(sample)

assert(libluaradio.luaradio_stop(radio) == 0)
assert(libluaradio.luaradio_wait(radio) == 0)
libluaradio.luaradio_free(radio)
```

closed 2020-04-02 by vsergeev

phrrngtn, 2020-04-02:
I'm afraid that your test script is failing for me even in-tree:

` (Pdb) /home/phrrngtn/test_luaradio_via_ctypes.py(50)<module>()
-> assert(libluaradio.luaradio_load(radio, script.format(snk_wr_fd).encode()) == 0)
(Pdb)
Segmentation fault
`
I will try on a different machine and architecture, get it working there and then try step by step on the Pi.

phrrngtn, 2020-04-02:
I built LuaJIT-2.0.5 from source and installed it to /usr/local. I then made sure that the C API to lua was working by doing a hello world type application from ctypes. I was able to get that working with Python 2 but not with Python 3. The Python program runs OK when I provide a full path to the luajit shared library in /usr/local. I got a radio example working with /usr/local/bin/luaradio and then it was fairly straightforward to transcribe it to a form compatible with your embedded example. There is only one libluaradio.so on my system so the .so name alone was sufficient.
```
lj = ctypes.CDLL("/usr/local/lib/libluajit-5.1.so", mode=ctypes.RTLD_GLOBAL)
#lj = ctypes.CDLL("libluajit-5.1.so", mode=ctypes.RTLD_GLOBAL)
# Load libluaradio
#libluaradio = ctypes.CDLL('/usr/local/lib/libluaradio.so')
libluaradio = ctypes.CDLL('libluaradio.so')
```

I was a bit surprised to see that the starting the script  takes about 6 seconds after loading. Is that to be expected?
```
phrrngtn@sdrpi:~ $ python ~/test_luaradio_via_ctypes.py
2020-04-02 14:29:59,934 - . - INFO - v0.6.0
2020-04-02 14:29:59,936 - . - INFO - called luaradio_new
2020-04-02 14:29:59,967 - . - INFO - loaded script
2020-04-02 14:31:06,081 - . - INFO - radio started
Found Rafael Micro R820T tuner
Exact sample rate is: 1102500.002464 Hz
Allocating 15 zero-copy buffers
2020-04-02 14:32:06,142 - . - INFO - woke up
2020-04-02 14:32:07,232 - . - INFO - stopped luaradio
2020-04-02 14:32:07,233 - . - INFO - waited on luaradio
2020-04-02 14:32:07,240 - . - INFO - freed luaradio
```

Here is the text of my example program:

```
script = """
return radio.CompositeBlock():connect(
  radio.RtlSdrSource(90.9e6 + -250e3, 1102500),
  radio.TunerBlock(-250e3, 200e3, 12),
  radio.FrequencyDiscriminatorBlock(1.25),
  radio.LowpassFilterBlock(128, 15e3),
  radio.FMDeemphasisFilterBlock(75e-6),
  radio.DownsamplerBlock(11),
  radio.WAVFileSink("/home/phrrngtn/wbur_mono_via_embed_2.wav", 1)
  )
"""

BTW, this still causes a segfault :-(
```
phrrngtn@sdrpi:~ $ luajit -e "require('radio')"
Segmentation fault
```
```

phrrngtn, 2020-04-03:
actually, it takes about 66 seconds (not 6!) between loading the script and starting the radio. It looks like the CPU is running at 100% for that time (I had thought that there may have been some kind of network-based timeout but one would expect CPU utilization to be close to zero for that)

vsergeev, 2020-04-03:
Long startup was due to FFTW plan initialization and seems to be limited to Debian/Raspbian. It's fixed in 7d52b1f of the devel branch, so feel free to give that a try.

I did try an install on Raspbian Buster with a Raspberry Pi 3 but haven't been able to see the segfault yet. May need to try Raspbian Buster on Raspberry Pi 4 next to try to reproduce more exactly.

phrrngtn, 2020-04-03:
I can confirm that 7d52b1f no longer has the long startup problem. Awesome!

phrrngtn, 2020-04-03:
... and the segfault is gone also!!
```
/usr/local/bin/luajit -e "require('radio')"
```

vsergeev, 2020-04-06:
That's great to hear, though I can't explain how that would fix the segfault you were seeing, as the FFTW plan initialization codepaths don't get executed with just the `require`. Let me know if you do run into it again and we can investigate further.

phrrngtn, 2020-04-06:
I think it was the 'build luajit from src' and then build luaradio against 
it that solved the segfault. In any case, it is good to be up and running. 
Thanks for your help.