vsergeev/luaradio #65: Parameter settings of my RTLSDR

choupette-change, 2020-09-09:
Hello everyone, I want to adjust the parameters (ppm, gain biastee, rf_gain ...) of my RTLSDR directly on the command line (examples / rtlsdr_rds.lua <FM radio frequency>, ...), is it possible ? if not can you find out how to do it differently? Thank you

vsergeev, 2020-09-10:
At the moment, you'd have to adjust them by modifying the script on the RtlSdrSource line. The source accepts a number of options in a table passed as the third argument -- a description can be found here: https://luaradio.io/docs/reference-manual.html#rtlsdrsource with some examples. But, the good news is that a built-in CLI frontend to all of the existing receivers is in the works.

There is a new release of LuaRadio coming soon (v0.8.0) with some fixes, new blocks, and a new receiver. After that, I'll be porting all of the examples into built-in applications in LuaRadio. You'll be able to specify the sources and sinks as well as additional options from the command-line. The additional options will probably take the form of a key-value string format you may have seen in other applications. For example,

```
./luaradio -a wbfm -i rtlsdr -o pulseaudio 91.1e6
./luaradio -a rds -i "rtlsdr,rf_gain=10,freq_correction=-1" -o json 91.1e6
```
where -a is application, -i is input, -o is output. Most applications will have reasonable default outputs, so you probably won't need to specify that every time. This will allow users to use all the existing LuaRadio receivers with other SDRs supported by LuaRadio, instead of hacking up the example scripts, and create a suite of usable applications out of the box.

choupette-change, 2020-09-10:
Thank you for your answer which suits me and works very well. Another question: how to change the color of Gnuplot curves in LuaRadio?

vsergeev, 2020-09-11:
ou can change the global gnuplot colors with the `~/.gnuplot` config file. There are many color schemes available for download here: https://github.com/Gnuplotting/gnuplot-palettes

vsergeev, 2020-09-11:
Closing issue for now. Feel free to open new ones if you have any more questions or run into problems. There's also a mailing list (still small): https://groups.io/g/luaradio

closed 2020-09-11 by vsergeev