vsergeev/luaradio #66: luaradio 0.9.0 osx build failure

chenrui333, 2021-01-05:
👋 trying to build the latest release, but run into some build issue. The error log is as below:

<details>
<summary>test failure</summary>

```
==> /usr/local/Cellar/luaradio/0.9.0/bin/luaradio test.lua
luajit: [string "radio.core.composite"]:0: dlsym(RTLD_DEFAULT, sigtimedwait): symbol not found
stack traceback:
	[C]: in function '__index'
	[string "radio.core.composite"]: in function '_reap'
	[string "radio.core.composite"]: in function 'wait'
	[string "radio.core.composite"]: in function 'run'
	test.lua:20: in main chunk
	[C]: in function 'dofile'
	/usr/local/Cellar/luaradio/0.9.0/bin/luaradio:110: in main chunk
	[C]: at 0x0102d24300
Hello, world!
Error: luaradio: failed
An exception occurred within a child process:
  Test::Unit::AssertionFailedError: <0> expected but was
<1>.
```

</details>

Full build log is in here, https://github.com/Homebrew/homebrew-core/runs/1644685048
relates to https://github.com/Homebrew/homebrew-core/pull/68254

chenrui333, 2021-01-05:
btw the 0.8.0 and 0.9.0 build, I did not find anything different (while 0.8.0 just builds and tested out fine).

Also I tried to update the test.lua to be as

```
      local radio = require('radio')

      radio.CompositeBlock():connect(
          radio.RtlSdrSource(88.5e6 - 250e3, 1102500), -- RTL-SDR source, offset-tuned to 88.5MHz-250kHz
          radio.TunerBlock(-250e3, 200e3, 5),          -- Translate -250 kHz, filter 200 kHz, decimate by 5
          radio.FrequencyDiscriminatorBlock(1.25),     -- Frequency demodulate with 1.25 modulation index
          radio.LowpassFilterBlock(128, 15e3),         -- Low-pass filter 15 kHz for L+R audio
          radio.FMDeemphasisFilterBlock(75e-6),        -- FM de-emphasis filter with 75 uS time constant
          radio.DownsamplerBlock(5),                   -- Downsample by 5
          radio.PulseAudioSink(1)                      -- Play to system audio with PulseAudio
      ):run()
```

got 

```
luajit: [string "radio.blocks.sinks.pulseaudio"]:0: PulseAudioSink: libpulse-simple not found. Is PulseAudio installed?
stack traceback:
```

chenrui333, 2021-01-05:
current `test.lua` file:

```
      local radio = require('radio')
      local PrintBytes = radio.block.factory("PrintBytes")
      function PrintBytes:instantiate()
          self:add_type_signature({radio.block.Input("in", radio.types.Byte)}, {})
      end
      function PrintBytes:process(c)
          for i = 0, c.length - 1 do
              io.write(string.char(c.data[i].value))
          end
      end
      local source = radio.RawFileSource("hello", radio.types.Byte, 1e6)
      local sink = PrintBytes()
      local top = radio.CompositeBlock()
      top:connect(source, sink)
      top:run()
```

vsergeev, 2021-01-05:
The formula is all good and the existing test script is a good test. You wouldn't want to use that front page example as the test script since it requires external libraries (librtlsdr and libpulse-simple) and doesn't run to completion. The problem appears to be that `sigtimedwait()` is unavailable on Mac OS X. I'll have to fix this with a workaround and release v0.9.1. Thanks for bringing this to my attention.

chenrui333, 2021-01-05:
Totally, that makes sense! Let me know when there is a next release, I am gonna close my PR for now.

chenrui333, 2021-01-08:
the patch works for me, thanks @vsergeev!

closed 2021-01-08 by vsergeev

vsergeev, 2021-01-08:
Thanks for testing it @chenrui333. I've released v0.9.1. The homebrew bump-formula-pr to 0.9.1 should work smoothly now.

commit 2a54b6cd (2021-01-06): core/composite: fix waiting for SIGCHLD with timeout on mac os x in _reap()

Mac OS X doesn't have sigtimedwait(), so we have to emulate it by
polling sigpending() in a loop.

resolves #66.
commit c86d7f9f (2021-01-08): core/composite: fix waiting for SIGCHLD with timeout on mac os x in _reap()

Mac OS X doesn't have sigtimedwait(), so we have to emulate it by
polling sigpending() in a loop.

resolves #66.