vsergeev/luaradio #76: Propagate exit code from block process to top-level process

vsergeev, 2022-08-27:
When a child block process exits due to a runtime error, the error is not propagated to the top-level process. The exit information is available from `waitpid()` in `CompositeBlock:_reap()`, and could be returned to `CompositeBlock:wait()`, and then returned to the user, so the user can decide how to handle the erroneous exit.

cdgraff, 2022-11-02:
Hi, @vsergeev I'll appreciate this improvement, are giving a lot of pain to manage restarts and failures without that.

Thanks!

vsergeev, 2022-11-08:
Hi @cdgraff, I have the change working locally. I just need to clean up some things and update the C API. I'll try to get a new release out this week.

vsergeev, 2022-11-11:
Hi @cdgraff, I've just released v0.11.0 with this improvement. CompositeBlock `wait()` and `stop()` now return a boolean indicating successful flowgraph exit. Let me know if that works for you and if you run into any other issues.

closed 2022-11-11 by vsergeev

cdgraff, 2022-11-11:
thanks! look good now!

```
11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:3] Starting process

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock] Connected output FrequencyTranslatorBlock.out to input LowpassFilterBlock.in

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock] Connected output LowpassFilterBlock.out to input DownsamplerBlock.in

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock] Aliased composite input TunerBlock.in to block input FrequencyTranslatorBlock.in

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock] Aliased composite output TunerBlock.out to block output DownsamplerBlock.out

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock] Connected output RtlSdrSource.out to input TunerBlock.in

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock] Connected output TunerBlock.out to input FrequencyDiscriminatorBlock.in

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock] Connected output FrequencyDiscriminatorBlock.out to input LowpassFilterBlock.in

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock] Connected output LowpassFilterBlock.out to input FMDeemphasisFilterBlock.in

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock] Connected output FMDeemphasisFilterBlock.out to input DownsamplerBlock.in

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] 

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock] Connected output DownsamplerBlock.out to input RealFileSink.in

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] 

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock] Flow graph:

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	RtlSdrSource [1102500 Hz]

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	    .out   [ComplexFloat32] -> {FrequencyTranslatorBlock.in}

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	FrequencyTranslatorBlock [1102500 Hz]

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	    .in    [ComplexFloat32] <- {RtlSdrSource.out}

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	    .out   [ComplexFloat32] -> {LowpassFilterBlock.in}

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	LowpassFilterBlock [1102500 Hz]

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	    .in    [ComplexFloat32] <- {FrequencyTranslatorBlock.out}

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	    .out   [ComplexFloat32] -> {DownsamplerBlock.in}

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	DownsamplerBlock [220500 Hz]

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	    .in    [ComplexFloat32] <- {LowpassFilterBlock.out}

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	    .out   [ComplexFloat32] -> {FrequencyDiscriminatorBlock.in}

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	FrequencyDiscriminatorBlock [220500 Hz]

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	    .in    [ComplexFloat32] <- {DownsamplerBlock.out}

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	    .out   [Float32] -> {LowpassFilterBlock.in}

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	LowpassFilterBlock [220500 Hz]

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	    .in    [Float32] <- {FrequencyDiscriminatorBlock.out}

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	    .out   [Float32] -> {FMDeemphasisFilterBlock.in}

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	FMDeemphasisFilterBlock [220500 Hz]

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	    .in    [Float32] <- {LowpassFilterBlock.out}

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [CompositeBlock]	    .out   [Float32] -> {DownsamplerBlock.in}
[... 93 lines cut ...]
11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] 

[... cut ...]
11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [DownsamplerBlock] Block pid 1218 terminating...

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] 

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] [RealFileSink] Block pid 1219 terminating...

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:4] 

11.11.22 11:17:29 (-0300) <encoder-fm-2> 2022/11/11 14:17:29 [input.external.rawaudio_0:3] Process exited with code 0

```

commit 518573c2 (2022-11-11): core/composite: return success boolean for wait() and stop()

resolves #76.
commit db33cc71 (2022-11-11): core/composite: add successful exit return value to wait() and stop()

resolves #76.